samsta/BatteryController #12: Reverse Relay Control Logic

NiallDarwin, 2021-08-31:
Change relay control logic from 'active high' **to 'active low'**
We are currently using 'active high' relays to control the battery contactors.
It seems that 'active low' is a much more common method and modules operating this way are much easier to source.

Note: Active high means the relay will switch on when voltage is applied to the control pin.
Active low means the relay will switch off when voltage is applied to the control pin.

Note 2: It's not really a bug but I don't know what else to label it.

NiallDarwin, 2021-09-06:
Figuring out how to achieve this change.

I believe the way to do this is in **testcan.cpp** I don't see any tests for this file and therefore I think I only need to dive in and change the program itself.

Reverse logic at lines 125, 126 (openBoth), 153, 159 (closeNegative and closePositive)

Change GPIO pins at lines 94, 95 This is necessary because we need gpios which pull high by default. Using Table 6-31 from the Pi's specsheet* I suggest pins 5 & 6 which is consistent with what I've done on James' python script.
* Page 102 of https://datasheets.raspberrypi.org/bcm2835/bcm2835-peripherals.pdf

samsta, 2021-09-06:
Yup, testcan.cpp is prototyping code and therefore doesn't have tests.

NiallDarwin, 2021-09-07:
Logic reversed and gpios changed in testcan.cpp

closed 2021-09-07 by NiallDarwin

samsta, 2021-09-07:
FYI, you can close issues by adding something like ‘closes #12’ in the commit message. This has the advantage of linking the commit with the issue and vice versa

NiallDarwin, 2021-09-07:
Good tip.