samsta/BatteryController #13: Automatically start software on Pi Powerup

NiallDarwin, 2022-01-25:
Not really an issue with the software but seems like this is a good place to pin the desire and keep track of progress.

NiallDarwin, 2022-01-25:
Also doing this may help close off [issue 10](https://github.com/samsta/BatteryController/issues/10).

NiallDarwin, 2022-01-25:
I think the best way to approach this is to:

1/ put the LBC power line back on a Pi controlled relay
2/ write a quick and dirty python program which gets called as part of the pi startup. The python program should startup the can networks then the battery controller program then switch the LBC on. 

Feedback on this idea please :)

JimsterCoder, 2022-01-25:
I have some experience trying to auto-start programs on Pi.  Starting something which runs and quits is fine but starting something that will run continuously is dodgy, or was 3 or 4 years ago when I looked into it. The problem is you can 'hang' the Pi and not be able to login to it, 'brick' it, essentially.  For this reason I don't do it with my home automation system.  I can review this again and see if anything has changed.  

I suggest modifying BC so that it starts the CAN connections and the LBC then commences its usual workings.

NiallDarwin, 2022-01-25:
> I suggest modifying BC so that it starts the CAN connections and the LBC then commences its usual workings.

I was trying to avoid this complication given that the Pi is a temporary solution. If you think it can be done in reasonable time then yes, feel free :)

samsta, 2022-01-25:
To autostart programs, Linux uses things like `systemd` which is available on the Pi. It runs programs as daemonized services, meaning they run in the background. As a consequence, you'll lose the ability to directly interact with them, for instance the curses-based interface won't work. Logs (anything written to standard output) go to a well defined location, though.

Have a look at https://www.thedigitalpictureframe.com/ultimate-guide-systemd-autostart-scripts-raspberry-pi/

Ultimately, we'll want to be able to run this as a background process and have other means to monitor it

JimsterCoder, 2022-01-25:
"other means to monitor it. It is a good solution."  Yes, I would like a web server interface.

I may be being optimistic but I think, for the short term, I can make the changes I mentioned above in a 'reasonable time'.  I'll look into it.

samsta, 2022-01-25:
> Yes, I would like a web server interface.

100% agree