samsta/BatteryController #16: Multi-battery startup sequence

JimsterCoder, 2022-10-24:
Preposed procedure for software startup sequence for multi-batteries.

Start comms with each/all batteries.

Wait for all batteries to enter 'safe to operate' mode
-if after 60 seconds a battery is not 'safe to operate', reboot it's LBC.
-restart 60 seconds delay
-if after 60 seconds a battery is still not 'safe to operate', open it's safety contactor
-send out appropriate alarm information

Allow Inverter to request closing of main contactors.
If there are batteries available and inverter requests closing:
-check voltages of batteries are within tolerance (of each other)
-if acceptable, close contactors

Monitor current flow to operational batteries to confirm closing of contactors.

samsta, 2022-10-25:
> 60 seconds

No idea if this is short or long compared to the normal/expected time to enter 'safe to operate'. What's the spec here?

> Monitor current flow to operational batteries to confirm closing of contactors

The presence of means the contactors must be closed, but the absence of current doesn't mean they're open. Something to consider from a safety perspective.

Other than that the proposal makes sense to me

JimsterCoder, 2022-10-27:
60 seconds is a long time, testing will reveal a more realistic time.  I fact it should take only a few seconds for each battery to declare itself 'safe to operate'.

Monitor current was my first thought on identifying if the contactors have closed.  Again, testing will hopefully reveal an practical method of confirming the contactors have closed/opened.

NiallDarwin, 2022-10-31:
Thanks for this James.
Timing
I agree we need to know the realistic timing before settling on a number. The LBCs are usually up and happy in a very short time-1 or 2 seconds max. What was the cause of the delay in operating on first Teensy test-is it relevant to our realistic time frame?

Number of Batteries
Can we have a way of auto-detecting batteries? EG if we build a 5 battery machine how will BatteryController know that it is a 5 battery machine and not a 6 whose 6th LBC isn't responding? Once the number of batteries is established a lack of communication should lead to disconnection of that battery as per your procedure.

Monitoring connection
The LBC should report voltage as well as current. We can use voltage to check if a battery is connected or not-a disconnected battery should have a more or less static voltage & zero current. This should be reliable except when the whole system is idling. As soon as the system starts moving energy it will become apparent.

Notes on Leaf behaviour
I got this from Walter at EVs Enhanced and could be useful for checking our main contactors are safe:
_The Leaf does some clever stuff with contactor control on startup which as I recall is turning on single relays at a time including the pre-charge one which should result in no voltage being received by the inverter as you should always need two relay before the inverter sees any voltage. So it is checking for a stuck relay as determined by voltage at the inverter at a time when there shouldn't be any. Then it does its normal pre-charge thing and voltage at the inverter needs to rise to close to its full value before the other relay kicks in and pre-charge is disabled._