samsta/BatteryController #19: Unlatch shunt relay if current is less than |1.5|

NiallDarwin, 2023-09-05:
Unlatch shunt relay if current is less than |1.5|
Currently when shunt is fired due to pack problem power remains on to the control relay.
Change so it stops powering the relay once pack current has dropped to less than |1.5|A

JimsterCoder, 2023-09-10:
Current limit set to 1.01A  Shut trip relay de-energized when battery current below this value.

This code needs to be tested.  TESTS need to be written for TeensyShutCtrl (and perhaps for TeensyRelay)

JimsterCoder, 2023-10-27:
Code needs to be re-thought. Has been removed.

NiallDarwin, 2023-10-29:
Can you be a little less cryptic so that when we return to it we know what it is about the code that needs to be rethought? Why wasn't it right?

JimsterCoder, 2023-10-29:
Code needs to be changed so the shunt status can be set for multiple reasons but the physical tripping of the shunt is only done in one place... and the reset done in one place.