samsta/BatteryController #22: Determine what to do if write to USB port (teensy) fails

JimsterCoder, 2023-11-08:
Currently when writing to the USB port fails a message (many!) are logged but no action is taken and, very oddly, the system seems to keep running, though it reports zeros in the log data but the invertered continues to function.  BC must still be sending messages to the inverter, perhaps with incorrect data. This would be bad as we count on this message to stop the inverter when the battery is full/empty.

JimsterCoder, 2023-11-08:
I suppose the simple answer, for now, is we need to shutdown or stop sending CAN messages to the inverter so that it shunts down.

JimsterCoder, 2023-11-08:
Thinking a bit more about this... BC kept on responding to the heartbeat from the inverter so the inverter was happy... despite the fact BC was not sending it any other data.  Nice, again, SinexHELL. 👎

NiallDarwin, 2023-11-12:
BatteryController should be able to detect this failure. It should then change its inverter messages containing max amps/power to zero.
We also need to flag this error to the end user. Where & how? In the 'finished product' do we think the operator will interact with battery controller as well as OEMS or OEMS alone?

JimsterCoder, 2023-12-13:
As we saw this morning, the teensy was unable to trip the shunts because the serial port connection had failed.  I have seen this failure approximately 4 times in the life of the project, the cause being unexplainable. Presently no action is taken by the battery controller but this is a serious issue so I will **modify the code to open the main contractors when any error relating to communications to the Teensy's occurs**.

NiallDarwin, 2023-12-13:
For contactor life & reliability it is much better to switch off under zero load.
If your shutdown can be staged to demand zero power from inverter then wait (5-10 seconds?) before opening the contactors that would be great. If you could do the reverse of startup that would be gold-plated goodness (Close pre-charge, open positive, open negative, open pre-charge).

JimsterCoder, 2025-08-04:
This was fixed on July 18,2025.  As a precaution. If ANY shunt gets tripped, which happens if USB comm are lost, the current limit is set to 0 then the contactors are opened.